m2i/MAVRIC-General #4: Raspberry Pi Zero HAT Size

Talonj123, 2017-06-14:
The size of the HAT may be too small for some of our electronics. See b48feaf
Options:
 - Use daughter cards to stack higher
 - Use separate cards to expand horizontally
 - Extend the edges of the PCB past the mounting holes
 - Save some space by going to smaller resistor packages
  - Increases chance of damaging the resistors while soldering

@matgyver Can you grab a quick look?
[image]

matgyver, 2017-06-14:
I wouldn't be too worried about damaging the smaller resistors.  Going to a larger board to give room is fine as long as it fits within any other mechanical constraints.  Be careful with stacking, if you are going to use the EEPROM for board identification, I don't think you will be able to stack them.

Talonj123, 2017-06-14:
I meant a specific pair, not multiple HATs. The whole stack would function
as a single HAT with 1 EEPROM.

Talonj123, 2017-06-14:
New Layout:
[image]
[image]

matgyver, 2017-06-15:
that looks good.

closed 2017-10-05 by GageLochner