m2i/MAVRIC-General #5: PCB Board Fabrication

Talonj123, 2017-06-14:
We need to be able to make fairly small runs of boards (say 10) preferably cheaply.
We might be able to use the same board for most of the HATs and just not populate unneeded parts (see #4). This would increase our volume, but also the chance that a revision would fail and need to be re-ordered.

 - What options do we have to manufacture boards?
 - Does M:2:I have any partnerships of university resources?
 - Does anyone have recommendations from personal experience?

 - Are there good ways to verify/test PCBs/designs before ordering?
   - How long does this take? (trade-off between budget and design cycle)
 - Automated simulation testing of PCBs? (is that a thing)

Options:
 - [Advance](http://www.4pcb.com/)
 - [OSHPark](https://oshpark.com/)

Talonj123, 2017-06-14:
@matgyver Does the university have any resources for fabricating boards?

matgyver, 2017-06-14:
@Talonj123 ECpE used to, but I'm not sure if it is still up and running.  M:2:I does have some limited capabilities using our CNC machine.

Talonj123, 2017-06-14:
Can M:2:I buy a desktop PCB mill?
 - [Prometheus](http://makezine.com/2015/11/16/first-look-prometheus-desktop-pcb-mill/)
 - [OtherMill](https://othermachine.co/othermill)

matgyver, 2017-06-15:
Maybe, that is something that is beyond this issue thread.  I will look into those options and go from there.

closed 2017-10-05 by GageLochner