m2i/MAVRIC-General #7: Raspberry Pi HAT EEPROM info

Talonj123, 2017-06-14:
Add links/other data as useful

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=108134

Talonj123, 2017-07-24:
At the moment, it looks like I can't set the state of an output pin. I can only set the pull state of an input pin. This seems like a serious limitation of the system. Regardless, the ability to self-identify to the board is nice. We can make a boot script that checks the which HAT is present and runs the correct `roslaunch` configuration on startup. This will let us get away with simpler updates (just pull the repository and compile) if we have 1 repository, and has no effect if we have separate repos.

Talonj123, 2017-07-24:
@matgyver Did you say you'd used these before at all? If so, did you explore having preset (high) outputs?

Talonj123, 2017-07-26:
https://www.kernel.org/doc/Documentation/devicetree/bindings/pinctrl/pinctrl-bindings.txt

Talonj123, 2017-07-26:
[This post](https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=188681) seems to indicate that not event the DT can configure the GPIO output states, I have a small amount of hope from the pinctrl link above, which has parameters to define `output-high` states.
I think the issue is that though the DT can define these states, it can't invoke them automatically or something.

matgyver, 2017-07-26:
I'm assuming you are wanting to set the default for the I/O pins.  Did you read this?  https://github.com/raspberrypi/hats/tree/master/eepromutils

Talonj123, 2017-07-26:
Yeah, but if I do ```setgpio 4 OUTPUT HIGH``` it does not output high. ```setgpio 4 INPUT HIGH``` causes it to pull the line up, but cannot drive say, an LED, and is just a bad idea for logic signals.

Talonj123, 2017-07-26:
https://www.raspberrypi.org/documentation/configuration/pin-configuration.md

matgyver, 2017-07-26:
So what is the concern here?  Do you want to make sure the pin is high or low on boot up?  If that was the case I would just use the weak pull up on the pi that can be configured through the EEPROM and setgpio or use a pull up resistor on the design of the hardware.  That would make sure your line is where it should be until pulled down by the pin.

closed 2017-10-05 by GageLochner