m2i/MAVRIC-General #16: Cross-cutting of milestones

Talonj123, 2017-08-07:
Often, milestones from one team require effort from another team to complete. For example, to make the drive system work, the electrical team needs the chassis from the mechanical team (at least partially). If the milestone is stated in such a way so as to remove this dependency (like running the system in simulation) then problems can be missed until too late.

Does anyone have thoughts on how to properly assign and track work on milestones that must cross team boundaries due to the integrated nature of the device?

Talonj123, 2017-08-07:
One concern here is that one team can essentially assign another team work, without going through that team lead for approval.

GageLochner, 2017-09-12:
All milestones will be worked out such that this issue does not arise. 
We will probably have a short meeting regarding this with the team leads on Saturday the 16th.

closed 2017-10-05 by GageLochner